valadaa48/retroroller #190: Retrolauncher fails after system update

Corentin-pro, 2020-11-09:
From fresh install (`archlinux-retroroller-20200613a.img`), if update is done (either ssh => `sudo pacman -Syu` or from GUI) the retrolauncher will fail to start (and because of being a service will loop on screen) : 

```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 567, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 884, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 775, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (humanize 3.1.0 (/usr/lib/python3.8/site-packages), Requirement.parse('humanize<0.6.0,>=0.5.1'), {'retrolauncher'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/retrolauncher", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3239, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3222, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3251, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 569, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 582, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 770, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'humanize<0.6.0,>=0.5.1' distribution was not found and is required by retrolauncher
```

Seeing http://tw.mirror.archlinuxarm.org/aarch64/community/ this should have started to happen since 2020 October 21th.

I couldn't find archive for ARM archlinux so I remove the package (`sudo pacman -Rdd python-humanize`) and intsall `python-pip` then `sudo pip install humanize==0.5.1` fixed the issue. I think this is a dirty workaround but this is working.

lethal-guitar, 2020-11-27:
Same problem for me. Your workaround fixed it for me as well, thanks @Corentin-pro!

valadaa48, 2020-12-13:
Closing as it seems there are two paths: 1) workaround posted here 2) Switch to RRVL

closed 2020-12-13 by valadaa48