valadaa48/retroroller #191: Buggy TIC80 core

klundry, 2020-12-04:
I was having bugginess with the tic80 core not launching carts. I recompiled a new one and its working as expected. Including my compiled core.
[tic80_libretro.zip](https://github.com/valadaa48/retroroller/files/5644070/tic80_libretro.zip)

valadaa48, 2020-12-11:
Thanks for rebuilding. I rebuilt the core in RRVL using our standard packaging and pushed it to the repo if you want to try to update in retroarch Online Updater.

I'm not sure what version you built, but I built from master and it seems to work well. Let me know if your version is better and provide a git-sha or I'll just upload your build.

valadaa48, 2020-12-11:
We made https://github.com/valadaa48/rrvl-cores live so in this case or in the future you can PR your updated core there.

klundry, 2020-12-11:
I just built it from master as well but I just tried out your rebuilt core from the online updater and it works great! Cheers!

valadaa48, 2020-12-11:
Awesome, great to hear!

closed 2020-12-11 by valadaa48