valadaa48/retroroller #192: Please consider to update some cores

raidenii, 2020-12-10:
As a continuation from Discord chat, here are the cores I compiled and the corresponding srcpkgs:

fbneo as of libretro/FBNeo@c093309
flycast as of libretro/flycast@b783996 (speedfix is libretro/flycast@3bc4830)
mgba as of libretro/mgba@ec030567
mednafen-ngp as of libretro/beetle-ngp-libretro@38f0e6a
ppsspp as of hrydgard/ppsspp@f9e7206

srcpkgs: [srcpkgs.zip](https://github.com/valadaa48/retroroller/files/5674194/srcpkgs.zip)
compiled cores: [cores.zip](https://github.com/valadaa48/retroroller/files/5674211/cores.zip)

valadaa48, 2020-12-11:
https://github.com/valadaa48/rrvl-cores is live if you want to do some PRs. flycast already updated.

valadaa48, 2020-12-11:
Closing this in favor of the new procedure

closed 2020-12-11 by valadaa48