valadaa48/retroroller #193: Updating breaks Retrolauncher

secondsol, 2020-12-11:
Traceback (most recent call last):
  File "/usr/bin/retrolauncher", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3239, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3222, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3251, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 567, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 884, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 770, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'retrolauncher==0.4.2' distribution was not found and is required by the application

Updating either through Pacman or the GUI triggers this. Tried futzing around with /usr/bin/retrolauncher and couldn't get anything working. Sorry, I'm not great with Linux so I'm not sure what else to try.

valadaa48, 2020-12-11:
Please check #190

secondsol, 2020-12-11:
Unfortunately that doesn't fix it. 

Again, I don't know this stuff so apologies, but it looks like it's looking for retrolauncher 0.4.2 only, and what's installed right now looks to be 0.4.2-2, or 0.4.2-4 in the testing repository.

I tried modifying the file to any variation of 'retrolauncher==0.4.2-2', or >=0.4.2, etc. but no dice.

valadaa48, 2020-12-11:
I'm not sure what the issue is here. Are you specifically tied to RetroLauncher? RRVL is the successor of RR ArchLinux if you don't mind booting into RetroArch directly.

secondsol, 2020-12-11:
I'm not specifically tied to it, but I do like having it functional. I'm not sure what the issue is since I don't really know Python, but it seems like a relevant issue, as RetroLauncher is one of the features that differentiates RR from RRVL, I think.

I guess I can set RetroArch to boot by default instead of RL, but it's a very nice and functional launcher that I'd like to still be able to use. For example, without RetroLauncher I'm not sure if there's a way to check that the overclocked kernel was properly switched out, and the CPU is running at 1.5Ghz instead of 1.3 in Performance mode.

The stock image from the download page works fine (although the global hotkeys are outdated; the hotkey image shows F2+directions for volume/brightness, but I believe in the stock, unupdated image, it's actually F3+directions), but I'd like to be able to run the latest version.

Is there any other info I can provide to try to nail down the issue?

valadaa48, 2020-12-13:
I wonder if it might be because python was updated to 3.9 from 3.8? Also, RR never released a proper overclock. If you check the RRVL wiki you'll find that we released proper overclock last night. I would still suggest you switch to RRVL as that is where all the new development is going.

secondsol, 2020-12-13:
I meant the 1.5ghz kernel that was posted in the original RR wiki, but I noticed the update in RRVL you mentioned so I switched over to that. No issues so far. Still not sure what the issue was, but oh well.

Can confirm that with the cpufreq set to 1512000, RA Flycast is REALLY solid now. Tested with Sonic Adventure 2, Street Fighter III 3rd Strike, Ikaruga, and MvC2, all are extremely playable with only minor slowdowns. Good stuff, thanks!

valadaa48, 2020-12-13:
I'm glad we solved your issue and you're on the latest RR distribution now. We put a lot of effort into RetroArch Spring which allows us to launch any external app from RetroArch itself. This effectively obsoleted RetroLauncher2.

That's great 1.5GHz is also working for you. RetroRun flycast may be a bit faster if you really want the better experience but will lose some functionality.

closed 2020-12-13 by valadaa48