valadaa48/retroroller #195: Question: RRVL is the fastest image - but can we add a nice frontend?

mtoensing, 2020-12-20:
Hi!

I tested nearly all Odroid Go Advance images: Recxalbox, Batocera, EmuELEC and this. The only thing that is obviously missing is a frontend. I like retroarch on my switch and I like the menus but on a dedicated handheld, I think a frontend is needed. 

So, is there an option do just do that somehow?

valadaa48, 2020-12-20:
`xbps-query -R -s emulation` will show emulationstation-go2. You can install it and if it works for you then you can use sv commands to make it your default.

The former RR ArchLinux ended up being a tangled mess of scripts and workarounds to support 3+ frontends. I decided to drop this support in RRVL since messing around with ES configs and fixes ended up taking up all of my time. 

That being said, while ES is popular, a lot of RRVL users are quite content with RetroArch. Further, there are 3-4 varieties of ES (EE/Batocera being the most fully featured) but aside from the ES go2 port by CrashOverride, the others rely on a ton of tightly integrated scripts that need customization for it to work outside of their respective environments.

If someone offers to maintain a particular ES everyone can agree on then I don't see any issue with it becoming a first class citizen. Personally, I would rather spend time on kernel, RetroArch, core and other optimizations.

valadaa48, 2020-12-20:
To be sure, if you decide to install emulationstation-go2 then you'll probably have to become familiar with and setup its relevant configs.

valadaa48, 2021-01-04:
Christian Haitian of ArkOS fame said he might help port his ES over to this

closed 2021-03-02 by valadaa48