valadaa48/retroroller #196: ppsspp core from online updater not working

klundry, 2020-12-20:
I've tried multiple isos and they all just kick me back out to the main menu when I try to use the ppsspp retroarch core. I know this was working before but not now. The isos I've tried all launch fine with the standalone ppsspp but I prefer to use the libretro cores if possible.

valadaa48, 2020-12-20:
I just added the libretro ppsspp core from EmuELEC. Let me know if that works. Also did you verify you put the bios in the right location?

klundry, 2020-12-20:
I have added the assets folder from the ppsspp repo to /home/odroid/.config/retroarch/system/PPSSPP/ as described in the libretro docs for ppsspp. Is that the correct location? Its still not working even after updating the core.

valadaa48, 2020-12-20:
System/bios directory was changed to /roms/bios - so it might be there you want PPSSPP.

Also, a new version of the core is there, I think the one from EE was crashing on startup.

klundry, 2020-12-20:
Ok, so would I create a PPSSPP folder inside /roms/bios and put the assets inside that?

valadaa48, 2020-12-20:
I think so, try it out

klundry, 2020-12-20:
Ok, I did that and its working now. Not sure if it was moving that folder or the new core that did it.

valadaa48, 2020-12-20:
OK great!

klundry, 2020-12-21:
I spoke too soon. Can we reopen this? It is working but not correctly. I can launch a single psp iso and it will work. If I close the game and try to open another psp iso, even the same exact one it will just show a black screen. I have to restart retroarch and then I can launch an iso successfully again.

valadaa48, 2020-12-24:
I see the same. It appears to have happen when trying to load state: https://github.com/hrydgard/ppsspp/issues/12841

raidenii, 2021-01-04:
@klundry The ppsspp core is reverted to a working version. Can you please test it on your end?

klundry, 2021-01-04:
@raidenii With the last update of RRVL and updating my cores I now can't launch any psp game with the retroarch ppsspp core and my network capabilities are gone. My previously working wifi dongle and usb ethernet adapter now do nothing.

klundry, 2021-01-04:
Not sure what caused my issues but I reflashed the latest image and installed the ppsspp core and it is working as it should.

klundry, 2021-01-06:
Alright I'm not sure what keeps causing this but its back to not working again. When it was working I hadn't put the psp assets folder on there yet and the save screens had no text but they functioned. I put the assets folder on and now isos won't load at all. I removed the PPSSPP assets folder and rebooted and they still don't work. I have not idea why this keeps happening. Other systems continue working normally.

klundry, 2021-01-06:
I'm trying to figure out the root cause but not having much luck. I enabled logging and I'm seeing some errors.

```
[libretro ERROR] [HLE] Unknown syscall: Module: (unknown) (module: 255 func: 4095)
[libretro ERROR] [HLE] Unknown syscall: Module: (unknown) (module: 255 func: 4095)
```

Also seeing this error. It says its not seeing that asset file but it is there in the folder. I've tried putting the PPSSPP assets in /roms/bios/PPSSPP and also in ~/.config/retroarch/system/PPSSPP.

```
[libretro ERROR] [SCEGE] Failed to load ppge_atlas.zim.

Place it in the directory "assets" under your PPSSPP directory.

PPGe stuff will not be drawn.
```

valadaa48, 2021-01-07:
@raidenii any ideas?

klundry, 2021-01-07:
I flashed a different micro sd from scratch again, did system updates, used the online updater to download the core and transferred some psp isos to the roms folder. And it works fine. I'm not sure where it gets messed up because thats how it was initially the two previous times as well. After initial setup it works fine, then after...something it stops working altogether even though I haven't done anything to mess with it. 

If  you can think of anything I should try or logs I should check etc. I'm all ears.

raidenii, 2021-01-07:
Honestly, I'm not not sure why it's recurring for you, plus I cannot reproduce the issue on my OGA. With that being said, the attachment is my PPSSPP asset (which is not up-to-date), but I never had issues with this.
[PPSSPP.zip](https://github.com/valadaa48/retroroller/files/5782761/PPSSPP.zip)

klundry, 2021-01-07:
Thanks. I don't think the assets are really the problem but I'll try it.

klundry, 2021-01-08:
I think I have narrowed down what is causing it. I started with a fresh flash again and all was working fine. Played a few different ISOs, all good. Then once I changed some of the PPSSPP core options by going to quick menu > options it crashes when I try to launch any PSP games now. It also seems to be wiping out other settings I've changed in retroarch as well. I like to change the menu input to flip the back and enter buttons and after it crashes that gets reset to the default.

valadaa48, 2021-01-09:
Seems like it's difficult to get this thing to work?

klundry, 2021-01-09:
Its not really difficult to get working now that I've figured out what has been causing my issues. Just have to make sure not to mess with any of the core options for now I guess. There are some pretty big bugs though.

closed 2021-03-02 by valadaa48