valadaa48/retroroller #197: Screen shuts off after 10 minutes even when being actively used

klundry, 2021-01-09:
This is on a fresh flash of the latest image, updates have been installed and roms copied over but no settings have been changed. After 10 minutes of being on the screen will turn off. The system is still running. Pushing buttons doesn't wake it up. The only way to get the screen back on that I've found is to tap the power button to sleep then tap it again to wake it up.

klundry, 2021-01-16:
Can anyone help point me in the right direction? Surely I can't be the only one that noticed this. I haven't touched any settings so this is just the default behavior of the image.

valadaa48, 2021-01-16:
I flashed the image from the downloads and let it sit for 30 mins but I wasn't able to reproduce it. This is on both OGS and OGA BE

klundry, 2021-01-16:
I have the OGA 1.0 but it seems odd that that would make any difference. I can flash a new sd and double check but what is managing the screen power?

klundry, 2021-01-17:
With some more testing to reproduce it I've discovered that it is related to sleep. After a fresh boot the screen will not shut off no matter how long it has been. Quick press of the power button to sleep > quick press to wake it up > 10 minutes later the screen will shut off.

valadaa48, 2021-01-17:
Thanks for narrowing it down. I'll see if I can reproduce it.

valadaa48, 2021-01-17:
I accidentally closed this. Reopening.

valadaa48, 2021-01-18:
I just pushed some updates, let me know if that helps after a reboot.

valadaa48, 2021-01-18:
I just left mine to sleep over night. I woke it back up and I'm idling in Retroarch in rgui using the oga driver. Will let it go 15 mins and see what happens.

valadaa48, 2021-01-18:
My device seems to be fine. Did you try changing the screen brightness? It might be worth checking in another OS that supports sleep like EE.

klundry, 2021-01-18:
Trying to change brightness does not wake it up. I just tried again with another sd that I flashed and did nothing to other than run updates, switch to oga/rgui. It may be related to actually running a core as well. After being woken up from sleep if I sit at the retroarch menu it will stay awake now. But, if I'm in a core/game even actively playing it will shut off after 10 mins.

valadaa48, 2021-01-19:
I flashed the image again, ran update, rebooted, retroarch changed to oga driver + rgui, launched a rom in nestopia and the screen hasn't turned off.

klundry, 2021-01-19:
Try this. Boot, sleep, wake up, sleep, wake up. I think it happens after sleeping twice.

klundry, 2021-01-19:
It happens every time after the second wake from sleep.

valadaa48, 2021-01-19:
I could reproduce it. I'll have to look into this.

valadaa48, 2021-01-19:
I might have narrowed it down to the kernel. I'll see if I can identify what setting introduced this issue.

valadaa48, 2021-01-20:
I just pushed a new kernel - can you update and reboot and let me know if it's fixed?

klundry, 2021-01-20:
No, it is still shutting off on me.

valadaa48, 2021-01-21:
Run `xbps-query -l | grep kernel` and make sure you have `odroid-go2-rrvl-kernel-4.4.189_11` and reboot in case? I just verified again that this kernel fixes the issue on my end.

klundry, 2021-01-21:
Checked my kernel and it is the version you mentioned. I rebooted again just to be sure. Started a game, sleep, wake, play for a couple minutes, sleep, wake, wait 10 minutes, screen shut off.

valadaa48, 2021-01-21:
if you run `sudo xbps-install odroid-go2-kernel` then reboot does it happen? This is the old stock kernel.

klundry, 2021-01-21:
Alright, did that and yes it still shuts off after 10 mins after the second sleep.

valadaa48, 2021-01-21:
I wonder if it's due to the upgraded u-boot? I think we should try to get more data (i.e, old RRVL images vs EE or some other image) to help narrow this down. Also, does it happen if you launch ppsspp standalone (so retroarch is stopped)?

klundry, 2021-01-21:
Just tried in standalone ppsspp and yes it still shuts off. I will try to flash an old rrvl image and not update and see what happens this evening.

klundry, 2021-01-23:
I flashed the 09-01-2020 rrvl image and the screen does not shut off.

valadaa48, 2021-01-23:
What happens if you do a full update on that flash?

valadaa48, 2021-01-23:
The reason I ask is because if after an update then this would prove that it's the OGS u-boot causing the issue.

klundry, 2021-01-23:
Full update as in "sudo xbps-install -Su"?

klundry, 2021-01-24:
I updated the system with the _rrvl/update.sh script and rebooted and tested again and the screen does not shut off. The u-boot is an older version than what is on my main card.

valadaa48, 2021-01-25:
Hmm, this is quite interesting indeed....

If you're 100% certain the old image with the old u-boot doesn't have this issue after updating using update.sh or even `/usr/bin/update` (same thing) then run /usr/bin/flash_uboot then reboot. This will merely just update uboot only to the OGS compatible u-boot that is in the latest image. If this is an issue then we should prob report this to HK on the forum.

klundry, 2021-01-25:
I will test again to be certain. Then update u-boot and test again and report back.

CEpeep, 2021-02-07:
I'm experiencing this issue as well. Running the latest 20210205 image on my OGS.

valadaa48, 2021-02-07:
This was traced to a needed change in boot.ini. Latest update + reboot will fix this once and for all!

closed 2021-02-07 by valadaa48