valadaa48/retroroller #198: Some problems on Odroid Go Advance

Ollibolli85, 2021-01-31:
Hello everybody,

i installed "RRVL 20210109 OGA 1.0, OGA 1.1, OGS" on my Odroid Go Advance and have the following problems:

There is no connection with a 2,4GHz-wifi-network, if the security is "WPA2 + WPA3"!
Just "WPA2 (CCMP)" works fine!

The Core "Nintendo - Nintendo (Murpen64Plus-Next)" do not start!
Other Cores works fine!

Many thanks in advance!

valadaa48, 2021-02-01:
Not sure who manages the mupen64plus core but I think most people use the standalone as it's faster.

Ollibolli85, 2021-02-01:
Thank you!

But there is no settings-/cheat-option with the standalone, right?

valadaa48, 2021-02-01:
Will let someone else chime in, I don't play it so really don't know.

valadaa48, 2021-02-06:
We removed support for mupen standalone. You can use either mupen64plus_next or parallel-n64 in the retroarch core updater.

closed 2021-02-06 by valadaa48