valadaa48/retroroller #199: mupen64plus standalone controller mappings

valadaa48, 2021-02-01:
`/usr/share/mupen64plus/InputAutoCfg.ini`

Here is a start:
```
[GO-Super Gamepad]
plugged = True
mouse = False
DPad R = button(11)
DPad L = button(10)
DPad D = button(9)
DPad U = button(8)
Start = button(15)
Z Trig = button(21)
B Button = button(0)
A Button = button(1)
C Button R = button(3)
C Button L = button(2)
C Button D = button(5)
C Button U = button(0)
R Trig = button(5)
L Trig = button(4)
Mempak switch = button(44)
Rumblepak switch = button(46)
X Axis = key(276,275)
Y Axis = key(273,274)
```

You might have to use a SDL test app to find the values. 

Also the button values seem to line up from the retroarch file except the analogs. You can check `/usr/share/libretro/autoconfig/udev/odroidgo3_joypad.cfg`

valadaa48, 2021-02-01:
Also for full screen you want to edit the ~/.config/mupen64plus/mupen64plus.cfg and change res from 320x240 to 640x480. If someone can test that res on the OGA BE as well we can just make it default.

tlayne, 2021-02-03:
> `/usr/share/mupen64plus/InputAutoCfg.ini`
>
> Here is a start:
>
> ```
> [GO-Super Gamepad]
> plugged = True
> mouse = False
> DPad R = button(11)
> DPad L = button(10)
> DPad D = button(9)
> DPad U = button(8)
> Start = [...]

I’m not sure this is the source of the button issue on OGS. I checked this file on both BE and Super and there is nothing in the file that indicates that there’s an Odroid joypad of any kind configured here for any of the versions. Oddly input works on BE and not super.

valadaa48, 2021-02-06:
We decided to not support this any more until someone can figure out how to configure it so that it works on all OGAs including OGS.

closed 2021-02-06 by valadaa48