valadaa48/retroroller #201: Entering Wi-Fi password that contains `$` does not save to config properly

cplr, 2021-02-01:
If you type in a Wi-Fi password containing the `$` character it is interpreted as a bash var (or something?). If you shut down and look at the NetworkManager config file, you'll see that the password is not properly serialized into the file.

Example:

passkey is `nHs$7nvJOfj)BEkd` and entered in using RetroArch's interface.

In the config file, you'll see it is saved as: `nHs)BEkd`

valadaa48, 2021-02-01:
Hard to say.. Here is the code: https://github.com/valadaa48/RetroArch/commit/f44162b8912cc05a6e9c9e6ca5c6f832877ade04

I also found this: https://stackoverflow.com/questions/40135708/escaping-dollar-sign-in-sh-c

if you're able to reproduce this in a quick popen(3) test and find a fix then we patch it.

Someone else had this issue as well and I think what they did was to create a new ssid with a password that didn't contain special chars.

valadaa48, 2021-02-06:
OK I recreated the issue and confirm that all special characters now work. I release a new base image in case others get stuck on new installs. If you don't want to flash, just update to the latest and your retroarch will be able to pair using your passphrase.

https://github.com/valadaa48/retroroller/releases/tag/20210205

closed 2021-02-06 by valadaa48

cplr, 2021-02-06:
Amazing, thanks!