valadaa48/retroroller #202: Trouble with TP-Link TL-WN725N v3, RTL8188EUS

cplr, 2021-02-01:
I just got the following Wi-Fi adapter, a [TP-Link TL-WN725N v3](https://www.tp-link.com/us/support/download/tl-wn725n/#Driver), which seems to use the `RTL8188EUS` chipset.

I am able to browse Wi-Fi networks, able to initially connect (on the Wi-Fi connect screen it says "Online") but going back out to check the network status, it's always on 127.0.0.1 and I am unable to do any network related activities.

Is this a symptom of a known issue? What can I do to help debug this?

On a Linux VM, this is how it shows up w/ `lsusb`:
`Bus 003 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter`

and with `dmesg`:
```
[ 4022.257042] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[ 4022.409059] usb 3-2: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[ 4022.409063] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4022.409065] usb 3-2: Product: 802.11n NIC
[ 4022.409067] usb 3-2: Manufacturer: Realtek
```

According to this issue: https://github.com/valadaa48/retroroller/issues/29 - the driver is included!

I am using the latest release (https://github.com/valadaa48/retroroller/releases/tag/20210109), and using an OGA v1.

cplr, 2021-02-01:
After some googling I found this: https://github.com/lwfinger/rtl8188eu

which says something interesting:

### NetworkManager does not list SSID
NetworkManager changes the Wi-Fi MAC address during scanning to improve privacy but this adapter does not support it. To adress this issue, please create `/etc/NetworkManager/conf.d/80-wifi.conf` with content:

```
[device]
wifi.scan-rand-mac-address=no
```

I will try that and report back here.

valadaa48, 2021-02-01:
Thanks, if you add that and it works then we can try to upstream it. I can test that it won't break my RTL8811AU but not sure about others...

valadaa48, 2021-02-01:
you can also rm /var/service/retroarch.service and reboot and get switch to tty2 with a usb keyboard and run ncmli directly to further troubleshoot.

cplr, 2021-02-02:
Hm, that did not seem to change anything for me. And I am now actually _reading what I pasted_ (astonishing I know) and it does in fact say "does not list SSID" which does not match up with my issue, which is that the connected state doesn't seem to "stay" beyond initially putting in the password..

valadaa48, 2021-02-02:
I did fix sleep a few hours ago, is it related to entering sleep mode?

cplr, 2021-02-02:
No, I have not been testing sleep at all. Thanks for the responses! Tomorrow I am going to try and build the driver from [TP-Link's page](https://www.tp-link.com/us/support/download/tl-wn725n/v3/).

valadaa48, 2021-02-03:
Also try running lsmod and see if the module is loaded.

valadaa48, 2021-02-06:
Not sure if it will help, but I updated the kernel if you want to give it a try.

cplr, 2021-02-06:
I will give it a shot as soon as I can. Cheers.

cplr, 2021-02-07:
YES the latest version works. I think it was just the special characters in WiFi password issue.

Thank you so much!

closed 2021-02-07 by cplr

cplr, 2021-02-07:
@valadaa48 with the latest update to the kernel, do you think there was anything else that could have been the fix for this, besides the special characters in the nmcli config issue?

The reason I ask is I had a similar issue with EmuELEC (could browse WiFi networks but not connect), and so that image might have a similar bug to the special character one. I'd love to be able to point the developer towards that issue for a possible fix there as well. 

I can't thank you enough- I can finally enjoy this wonderful image to its fullest extent.