valadaa48/retroroller #205: alsa-output switching

nightflasher, 2021-02-08:
yesterday i've plugged my headphones into my odroid go advance BE and everything was fine, after unplugging the sound didn't came back on the speaker. today i've ssh'ed into the rrvl and went straight to the alsamixer and set the "Playback Path" back to SPK_HP. 

[code]
asoc-simple-card rk817-sound: rk817-hifi <-> ff070000.i2s mapping ok
asoc-simple-card rk817-sound: ASoC: no sink widget found for MIC_IN
asoc-simple-card rk817-sound: ASoC: Failed to add route Mic Jack -> direct -> MIC_IN
asoc-simple-card rk817-sound: ASoC: no source widget found for HPOL
asoc-simple-card rk817-sound: ASoC: Failed to add route HPOL -> direct -> Headphone Jack
asoc-simple-card rk817-sound: ASoC: no source widget found for HPOR
asoc-simple-card rk817-sound: ASoC: Failed to add route HPOR -> direct -> Headphone Jack
input: rockchip,rk817-codec Headphones as /devices/platform/rk817-sound/sound/card0/input1
rk808-rtc rk808-rtc: setting system clock to 2021-02-08 12:38:42 UTC (1612787922)
#0: rockchip,rk817-codec
[/code]

nightflasher, 2021-02-09:
After having no sounds at all, I've switched the sdcard (hardkernel provided image) to check if there's any hardware issues. I had sound, switched sdcard again, ran the rrvl_reset script and sound was back again. 
After checking the retroarch.cfg with the one which seems to be faulty, I couldn't find any difference.

What could happen here?

valadaa48, 2021-03-02:
Is this still an issue after updating?

valadaa48, 2021-03-03:
Closing due to no response and we haven't heard of it happening to others

closed 2021-03-03 by valadaa48