valadaa48/retroroller #206: Question: Does retroroller provide a full void linux with it?

brandflake11, 2021-02-08:
This is not an issue, just a question trying to figure out what retroroller is:

Does retroroller provide a full void linux underneath of it? LIke, can i ssh into an ODroid Super with it and install emacs to get work done? I ask because images like Batocera don't provide a full linux distro that can get work done too, and I want to find a gaming distro for the ODroid that can also allow tinkering with Linux.

Thanks for any help you can offer.

valadaa48, 2021-02-08:
Yup, it's a full Void Linux distribution with just a few custom packages. The rest of the Void packages are installable.

brandflake11, 2021-02-09:
Oh my, that is awesome! One last question: is kodi easy to install on retroroller?

valadaa48, 2021-02-09:
I think Kodi needs Xorg so you'd have to figure out to install Xorg and get it running. There's another image based on buildroot that does it but I don't think they released source or instructions unfortunately. You could try with this distro and ask meveric on the forum how he/she did it.

brandflake11, 2021-02-09:
Okay, thanks so much for everything! I am excited to try this out once I get the ODroid Go Super!

closed 2021-02-09 by brandflake11