valadaa48/retroroller #207: Charging when Wifi Fi is connected

gmalarkey, 2021-02-09:
Handheld: Odroid Go Super

 I usually have WiFi on and connected for Retro Achievements. If I try to plug in and charge my OGS while it is on and WiFi is on/connected, plug it in while it is on and WiFi is on/connected and shut down, plug it in while it is on and WiFi is on/connected and put in sleep mode it doesn't charge. I either need to unplug my WiFi adapter and reboot then charge or use the hot key to go into dark mode then charge.

valadaa48, 2021-02-10:
What do you think can be done here? I don't think it's RRVL specific unless we need a kernel fix, uboot upgrade etc..

Aldenwil, 2021-02-10:
I don't believe it's RRVL specific either, but possibly killing Wireless when sleep is triggered via the power button may allow for charging - but it does seem to be a device issue itself, not specific to this image.

gmalarkey, 2021-02-10:
To be honest I wasn't sure what the next steps are, I was hoping that we could maybe determine what the root cause was or get a better idea of what it might be. Then if needed submit the issue with some documentation to HardKernal. If I can help do some sort of testing I'd be happy to.

valadaa48, 2021-02-10:
The problem seems to be even for me that wifi draws too much power and wifi appears to be drawing power when suspended. If you run ogainfo in ssh while charging and wifi is inserted you'll see that the charging amount is negative or very small positive. If you reduce screen brightness all the way down then it charges faster.

gmalarkey, 2021-02-10:
So is that what we should post on the HardKernal forums? Maybe there's a fix they can implement to stop WiFi from drawing when suspended? It seems that the power it's pull vs receiving is a hardware issue right?

valadaa48, 2021-03-02:
Closing this as it's not RRVL specific

closed 2021-03-02 by valadaa48