valadaa48/retroroller #209: OGA1.0 can not CPU Overclock

lancherliu, 2021-02-20:
I am a Chinese，i hope you can understand what i said.
My device is OGA1.0，I Edit /boot/rrvl.conf and uncomment the cpufreq=151200，then In SSH the CPU stats CPU Freq is 100800.
And i I Edit /boot/rrvl.conf and uncomment the cpufreq=124800，then In SSH the CPU stats CPU Freq is 124800.
Anything what i do,it can not overlock to 151200.
[image]
[image]

valadaa48, 2021-02-21:
I pushed a new kernel update out that should fix this. I would suggest you boot without hte OC and run it manually using setfreq to make sure your unit can handle it

lancherliu, 2021-02-22:
yep,it is 151200,thank you @valadaa48 
[image]

closed 2021-02-23 by valadaa48