valadaa48/retroroller #210: F1 Hotkeys Don't Work Odroid Go Super

brandflake11, 2021-03-03:
### System

* xuname:  Void 4.4.189_RRVL-23 aarch64 Unknown uptodate rFFFFF
### Expected behavior
Use the F1+F2 to access the Retroarch menu. Use F1+A to toggle fast forward. Use F1+Y to see FPS
### Actual behavior
Nothing seems to happen
### Steps to reproduce the behavior
1. Load a game in Retroarch
2. Try a shortcut above
3. Notice it isn't working on Odroid Go Super

Is this happening because the documentation is not up to date, or is this a bug?

Thank you for your time!

valadaa48, 2021-03-04:
I think the wiki needs to be updated. Since the release of the OGS, we had no choice but to unmap these retroarch "hotkey" controls since F1 is different on OGS and v1.1... I agree it's annoying but getting this right out of the box for all platforms would need considerable config, testing and maybe scripting - then the issue happens when people swap their sdcards between devices...

valadaa48, 2021-03-04:
I updated it, you'll need to manually map them as I described. Closing the issue.

closed 2021-03-04 by valadaa48

brandflake11, 2021-03-04:
Thank you for the clarification. Where are these keys set? Is it in ~/.config/retroarch/retroarch.cfg?