valadaa48/retroroller #211: Genesis Plus GX Wide Core

brandflake11, 2021-03-09:
Hello,
Is it possible to get [Genesis Plus GX Wide](https://www.libretro.com/index.php/genesis-plus-gx-wide-now-available-for-libretroretroarch/) on retroarch for the Odroid Go Super? I think this core would work great on the Super, if it is able to be made on arm and isn't too intensive.

Also, another question, do you specifically maintain the cores for the RRVL, or is it pulled from the main Retroarch team?

Thanks for your help!

valadaa48, 2021-03-10:
The core is there, just the core info is missing https://github.com/christianhaitian/retroarch-cores/tree/master/aarch64

closed 2021-03-10 by valadaa48

brandflake11, 2021-03-10:
Thank you, I take it the retroarch cores are taken from this repository then.